firmwaremodules/stm32-secure-patching-bootloader #8: STM32H7 support

magnifikus, 2023-03-10:
Any plans for this?

Im using the H725 with an external qspi flash (large app with http data), so im very interested in helping here.
One problem i see, is the 128kb sector size (minimal erase)

Would appreciate hints and suggestions.

firmwareguru, 2023-03-10:
As a matter of fact I've recently completed a port to the H7 platform and specifically the DISCO-H745I.  I'll be rolling this board into the list of new platforms supported for the v1.4.0 release - around the end of this month.

The H7's uniform 128 KB sector size isn't really a problem - unless you're trying to use internal flash to store different sets of data that don't necessarily line up with 128 KB and you're using a small device like the 725 (512 KB model).

The bootloader is going to take the first sector (128 KB).  On a 512 KB device, that would leave 3 for the application - kind of hard to split that in half for a dual-slot internal flash layout.  You could, as you've suggested, use external flash configured for memory mapped access mode to extend the active slot (SLOT0) past the 3 sectors of internal flash to cover additional sectors on external flash.  You wouldn't want to necessarily run the executable portion of your app from QSPI, but it would be possible.

More traditionally the internal remaining 3 sectors (384 KB) could get assigned to SLOT0 and an equivalent amount in QSPI flash is assigned to the download SLOT1.  

Either way, the stm32-secure-patching-bootloader has you covered ;)

magnifikus, 2023-03-22:
after some discussions we sadly have to skip, the pricing for a binary blob is abit off sorry.

closed 2023-03-22 by magnifikus

firmwareguru, 2023-03-23:
Thank you for at least considering it.  You can still use it on one of the supported dev boards to implement a prototype.  Best of luck.

magnifikus, 2023-04-13:
thank you for the fast reply and the offer!