firmwaremodules/stm32-secure-patching-bootloader #9: STM32H745I support

dolence, 2023-05-03:
Hi, Evan! How are you?

I just got an Riverdi display and MCU board featuring the STM32H757XIH6. It looks like a nice board with a good amount of external RAM e QSPI flash avaliable and TouchGFX support. I'm still waiting for it to arrive and saw your project while browsing for how to develop a bootloader capable of updating firmware using a thumb stick.
Is there any chance to have it supported? If not, would be possible for the common user to adapt the existing STM32H745I example to STM32H757X or is it a paid feature only?

firmwareguru, 2023-05-03:
Hi @dolence,

Short answer is yes, the stm32-secure-patching-bootloader can be engineered to support a project using a Riverdi board.

As you know I'm only supporting the development boards for STM32 - Nucleo, Discovery mostly, and SeeedStudio for the WL's,  on this GitHub repository for no cost.  The intent is to get a product developer up and running and see if the solution is a fit for an eventual commercial offering (The Nucleo & Disco boards come with licenses that restrict use in commercial products).

The Riverdi boards are commercial-ready, and require a customized stm32-secure-patching-bootloader.  Different flash chip, USB pinouts, interfaces, etc.  Also the eventual installation location of the board will dictate some design decisions.  

If/when you want to start a project, you know where to find me :)

dolence, 2023-05-03:
I really crossed with your project a few hours ago. I didn't even knew about the commercial restrictions you have mentioned. This project is on a very early stage, and I've choose Riverdi because of the big 10.1" high res display and MCU/additional ICs. I do have another H7 board but it's from Waveshare  and I had to do a lot of wiring to have it working with a much smaller display from another brand.
We are a very small familiar business. It's only me and my uncle working together. I expect we grown up to be partners someday, but at this moment all I have left is to try to implement some simpler solution by myself.

Thank you for clearing that up. I hope to talk to you soon.

firmwareguru, 2023-11-01:
DISCO-H745I is supported in v1.4.0.

closed 2023-11-01 by firmwareguru