firmwaremodules/stm32-secure-patching-bootloader #10: Support for new board

jkoay, 2024-02-29:
Can you please add support for NUCLEO-H563ZI board?

firmwareguru, 2024-03-09:
Thanks for your interest @jkoay.  I don't have any spare cycles at the moment to look at this, but I can add it to list of possible boards to support for the bootloader 1.5 release.  

The H563ZI is a cortex-M33 with TrustZone.  Just FYI, the stm32-secure-patching-bootloader does not leverage TrustZone features.  Root of trust, however, would be established the same was as say, a TF-M solution, using RDP Level 1 or 2.  Other than that, the H563ZI would work like most other boards.  

If you are you using external flash / displays / TouchGFx this is a good solution for that.

harshrmistry-eaton, 2025-01-13:
Any plans to add support for H743 and U575 running mcuboot? I am interested to know if this works with Zephyr RTOS. If not, any plans to port this to Zephyr? Please share the road map.