firmwaremodules/stm32-secure-patching-bootloader #11: Support for Zephyr RTOS

harshrmistry-eaton, 2025-01-13:
Hi, any plans to add support for H743 and U575 running mcuboot? I am interested to know if this works with Zephyr RTOS (preferably 3.7.0 LTS). If not, any plans to port this to Zephyr? Please share the road map.

harshrmistry-eaton, 2025-01-16:
@firmwareguru Any update on this?

firmwareguru, 2025-01-16:
Hi there, there are no plans at this time to add any more features or supported boards or OSs to this project.  It has achieved a certain level of maturity and stability that makes it useful for my purposes, namely on the WLE5 MCUs.

For Cortex-M33 class of MCUs like the U575 I would recommend ST's TF-M solution port which as you know uses mcuboot under the hood.  I'm not sure where that stacks up with Zephyr however.